Yona-Appletree/LEDscape #23: Minor Service and Install Suggestions

cwshep, 2015-01-06:
It seems that in ledscape.service.in, it may be better to automatically restart the service if it dies, as well as set the niceness a bit lower (so it has higher priority), e.g.:

Restart=always
Nice=-20

Also, I had issues with setting my network configuration through /etc/network/interfaces on the current BBB image.  Essentially it worked on boot, but as soon as the network cable was disconnected the interface lost its static IP.  Apparently this is due to wicd.  Thus you can either uninstall wicd, or set the IP via wicd-curses (or the appropriate config file in /etc/wicd/).

RGB-123, 2015-01-08:
I am going to be doing a write up on how to properly setup a static ip on the wireless side of the BBB.
Basically I modified things  using vi as issued below and updating the interfaces script
-------------------------------------------------------------------------------------------
vi /etc/network/interfaces
-------------------------------------------------------------------------------------------
# WiFi Example
auto ra0
#iface ra0 inet dhcp
#    wpa-psk  "#################"
#    wpa-ssid "#################"
 
iface ra0 inet static
address 192.168.1.118
netmask 255.255.255.0
gateway 192.168.1.1
    wpa-ssid "##########"
    wpa-psk  "##########"
dns-nameservers 8.8.8.8 192.168.1.1
 
# Ethernet/RNDIS gadget (g_ether)
# ... or on host side, usbnet and random hwaddr
# Note on some boards, usb0 is automaticly setup with an init script
iface usb0 inet static
    address 192.168.7.2
    netmask 255.255.255.0
    network 192.168.7.0
    gateway 192.168.7.1

GGreenwood, 2015-01-08:
I've had success either by setting the static IP via wicd-curses, or by letting it use DHCP and setting my router to assign a static address to a given MAC address or hostname.

cwshep, 2015-01-12:
Yes, my main point (for static IPs) was that wicd interferes with disconnecting and reconnecting the ethernet cable (it erases the static configuration), which is not something clear or easy to debug.  Some of the straightforward solutions are 1) remove wicd, 2) use wicd-curses, or 3) use dhcp and set the IP via the DHCP server.