Yona-Appletree/LEDscape #25: Slim (console only) Debian build for Beaglebone Black that works with LEDscape?

seriousmumbo, 2015-01-27:
https://groups.google.com/d/msg/beagleboard/GANFPr7BtVo/ubCbAW2cSysJ

Hi, thought I'd cross post here as I imagine some of you have the same issue.  Anyone here found a slim console only image with Debian (or Ubuntu) that plays well with LEDscape?

RGB-123, 2015-01-27:
Ryan, it sounds like you may have worked this out.  However, I think you simply needed to disable the HDMI in the uEnv.txt file.

seriousmumbo, 2015-02-05:
Yes, kernel bone70 works with HDMI disabled.  I'm curious as to why this is necessary?  I like the HDMI =).

mykolasmith, 2015-02-05:
It's because LEDscape uses those GPIO pins that would otherwise be used for HDMI. You can't have your cake and eat it too ;)

seriousmumbo, 2015-02-05:
Speaking of tradeoffs: I noticed that the ledscape dtb breaks sd card hotplug.

seriousmumbo, 2015-02-05:
Is the .dts source for the dtb available?

seriousmumbo, 2015-02-06:
I created a new dtb, it should work well with the newer kernels (tested with bone70).  Advantages include: sdcard hotpug working, hwrng, enhanced quadrature encoder pulse decoder, bandgap (chip temp) sensor, GPS cape and much much more:

https://github.com/seriousmumbo/ledscape-dtb

My use case (one 24 pixel strip) is probably more limited than yours, so testing is in order.