Yona-Appletree/LEDscape #27: FPS Limit

jnordberg, 2015-03-11:
I'm using LEDscape to drive just a handful of ws281x LEDs and I'm seeing a CPU usage of about 40%. Is it possible to limit the fps? I don't really need 3k updates per second :)

```
[render] fps_info={frame_avg_usec: 301, possible_fps: 3322.26, actual_fps: 3185.30, sample_frames: 31853}
```

RGB-123, 2015-03-11:
j, what does your json config look like?

jnordberg, 2015-03-11:
```json
{
	"outputMode": "ws281x",
	"outputMapping": "rgb-123-v2",
	"demoMode": "fade",
	"ledsPerStrip": 4,
	"usedStripCount": 1,
	"colorChannelOrder": "RGB",
	"opcTcpPort": 7890,
	"opcUdpPort": 7890,
	"enableInterpolation": true,
	"enableDithering": true,
	"enableLookupTable": true,
	"lumCurvePower": 2.0000,
	"whitePoint": {
		"red": 0.9000,
		"green": 1.0000,
		"blue": 1.0000
	}
}
```

RGB-123, 2015-03-11:
Oh wow, you are only running a few strips.  You may want to turn off dithering and interpolation by setting these to "false"

jnordberg, 2015-03-11:
Yeah I realise that my usage isn't what this lib was written for :) Setting interpolation and dithering had no effect, but I found that increasing the number of LEDs per strip did, managed to get it around 10% with 64.

Yona-Appletree, 2015-06-10:
You're absolutely right that this sort of small setup isn't what LEDscape is currently optimized for, but I think it's a good direction to move in. I would welcome and pull requests for this sort of functionality.