Yona-Appletree/LEDscape #37: LED display unstable during slow crossfades

mazo123, 2015-10-25:
Hi Yona, I'm having some stability issues while trying to create a smooth, slow changing output display. I have tried turning off dithering and interpolation but the output is not smooth at all. I have attached a simple processing sketch that performs single stepping of the RGB values and on my system, the instability is quite obvious. I'm running a Beagle Bone Black and I have updated it to the latest build and I'm running Processing from a PC. My processing code is below and I'm using an OPC.pde file which says it was modified by you in late 2014.

Any suggestions? 
Thanks for you help.
Paul

/**
 * This program single steps the RGB values to test the stability of the output  
 */ 
  
OPC opc;
int x;

//========== Setup ===========================
void setup() {
  size(400,400);                
  opc = new OPC(this, "192.168.0.25", 7890); 
   float spacing = height / 16.0; 
   opc.ledGrid16x16(0, width/2, height/2, spacing, 0, true);     
}

//=============== Draw ====================================
void draw() {                     
       
  x++;
  if (x == 155) {
      x = 0;}
      
  println("red = ",x, "Green = ",x, "Blue = ",x/2);   
  loadPixels();
  
    for (int i = 0; i < (width*height); i++) {        
      pixels[i] = color(x,x,x/2);
    }
  
  updatePixels();      
  delay(500);   
  opc.writePixels();       
}

RGB-123, 2015-10-25:
Paul, are you physically connecting to the beagle bone black from your PC? If it is over Wi-Fi there can be significant propagation delay.

Respectfully,

Ryan O'Hara, PhD
RGB-123
18 Larchmere Dr
Beavercreek, OH 45440
937-430-8516
sales@rgb-123.com<mailto:sales@rgb-123.com>

mazo123, 2015-10-25:
Ryan,
The PC and the BB are both connected to my router via Ethernet cable so I doubt that this is the issue. This is the same instability problem I have been trying to resolve for several months that I spoke to you about originally. I thought I'd give it one last try.

GGreenwood, 2015-10-25:
Do you have a resistor in between the beaglebone and the first LED?

Yona-Appletree, 2015-10-25:
Mazo123, sorry that you're having trouble with LEDscape. Looking at your setup, I have a few questions and ideas:

1) You have a delay(500) in the Processing sketch. This means you're only sending new data to the BBB every half-second, at best. With interpolation, this may look OK, but it is generally much better to send your data at a higher framerate, like 30 or 60fps. Try chaning the delay to 33 milliseconds.
2) How do things look with LEDscape's demo mode? By default, you should get a rainbow fade on your pixels.
3) Can you give us the command line arguments you're using to start opc-server? That may also help.
4) How long are the wires connecting the BBB to the LEDs, and what cape, level shifter or other hardware exist between the BBB and the LEDs?

Thanks, I hope we can figure it out.

- Yona

mazo123, 2015-10-26:
Thanks for the advice. Here are some answers:

1. I tried increasing the frame rate and that helped a bit but the display is still not smooth
2. The demo mode looks great. It has nice smooth colour fades which is what I am trying to replicate albeit slower
3. In terms of the OPC server, I use the following statement in my startup.sh file:
sudo ./opc-server --config ws281x-config.json

and here is my json

{
        "outputMode": "ws281x",
        "outputMapping": "rgb-123-v2",
        "demoMode": "fade",
        "ledsPerStrip": 256,
        "usedStripCount": 48,
        "colorChannelOrder": "BRG",
        "opcTcpPort": 7890,
        "opcUdpPort": 7890,
        "enableInterpolation": true,
        "enableDithering": true,
        "enableLookupTable": true,
        "lumCurvePower": 2.0000,
        "whitePoint": {
                "red": 0.9000,
                "green": 1.0000,
                "blue": 1.0000
        }
}

4. In terms of hardware, I have an RGB123 BB 24 output cape which takes care of the level shifting and I have a 220 ohm series resistor between the BB cape and the LEDs. The signal wire from the BB to the LEDs is about a foot long 

I'll get Ryan to email you my full Processing sketch so you can see what I am trying to do as it is too long to paste in here. All I'm trying to achieve a smooth slow changing display but I'm at a bit of a loss as to why I can't get it to look smooth even with interpolation and dithering on.

Thanks for your help.
Paul

Yona-Appletree, 2015-10-26:
Well, the bright side is that if demo mode looks good, it's probably a very solvable problem, and we really don't have to worry about your hardware. I'd love to look at your processing sketch, it probably has something to do with how the data is being generated / sent or something specific to your network. One thing to try is to generate the OPC data from a script (node or python, probably) on the BBB itself, to account for network issues.

Yona-Appletree, 2015-11-02:
Just wanted to see if you had any luck with this. Let me know if you need anything more, I'd like to close this issue (and have your issue solved) soon.

mazo123, 2015-11-03:
Yona, did you receive my Processing sketch from Ryan? If not, how can I send it to you? 
And in order to do some further testing, would you be able to post an example OPC python program that I could run on the BBB? I'm rather new to Linux etc.