Yona-Appletree/LEDscape #38: WS2811 not working

Summer88, 2015-10-26:
Hi,
I've tried the defaul config  "ws281x" with a strip of 5vdc WS2812B and everything was fine.
Then I've replaced the strip with a 12Vdc  WS2811 with one chip driving 3 led and the strip turns on but doesn't respond correctly to signals.

this is how I've wired the WS2811:

12vdc power supply:   
                  ( + ) ---------------- STRIP (+)
                  ( - ) -----------------STRIP (-)
                  ( DI ) ---------------- level shifter (H1)
5vdc power supply:
                  ( + ) ---------------- level shifter (H)
                  ( - ) ----------------- com (-)
3V3 BEAGLEBONE ---------- level shifter (L)
CHANNEL0 PIN  ---------------level shifter (L1)

every help is really appreciated, BR

Yona-Appletree, 2015-10-26:
I've seen those modules require different voltages for signal. Try a 5v signal. Other than that, it's hard for us to help because it sounds like LEDscape is working correctly for your WS2812Bs, and that the problem with the other modules is specific to them.

Summer88, 2015-10-27:
Thanks for the help, I've already tried the  signal at 5v with the level shifter. With raspberry no problem in driving them using rpi_ws281x library.

Yona-Appletree, 2015-12-31:
LEDscape only supports 800mhz ws2811, unfortunately. I personally have never encountered one of the slower ones, nor have we received any requests for support for them (other than this). It wouldn't be very hard to add support, however. Can you verify that your LEDs are of the slower sort by checking the parameters of rpi_ws281x or by using a logic analyzer? Thanks.

jammerxd, 2021-01-09:
I too am having problems with WS2811 strips. I also have a WS2815 strip that I tried and neither of them functioned with LEDscape. Can you make specific timing templates for them instead of ws281x.p make ws2811.p, ws2812B.p, ws2815.p, etc...? I'm not even sure how to adjust the timings based on the data sheets because I don't know enough about your structure to make those changes. Where is the low time defined? Where is the high time defined? Where is the reset time defined? I can't pick out what's going on in the .p file to adjust the ws281x.p to my needs.

bigjosh, 2021-01-11:
I know the PRU code well and can make timing adjustments for you if necessary, but first let's make sure that is the problem. 

Are you using a 5 volt level shifter to feed the data in signal of the strips? It is not always necessary, but some strips require it.

Do you have an Arduino handy that you could use to test these strips?

jammerxd, 2021-01-11:
Hi @bigjosh, 

Thanks for the answer, I've got an arduino and when I use fastLED and set the LED type accordingly, the strip works fine(using WS2811 timing for the WS2815 has been fine - WS2812 and WS2812B timing for the WS2815 strip produces the same result I see on the beaglebone).

I'm not sure if a logic level shifter is required for the WS2811. I've got the WS2815 strip that doesn't require it when hooked up to the arduino. I'm using an arduino UNO for reference.

I haven't tried the WS2811 on the arduino yet because I can't determine if the WS2811 needs 5v or 3.3v for DI and if the arduino UNO digital out is 3.3V or 5V.

Also to clarify on the strips I have:

1xWS2815 - 300 Pixels
2xWS2811 - 160 Pixels ea.

bigjosh, 2021-01-11:
UNO digital out is 5V and WS2811 uses 5V for data lines.  Almost all strip types I've seen use 5V for data lines. Some strips will work connect directly to BB's 3.3v data lines, but not all. So things that could be wrong...

1. Bad strip
1. Voltage level mismatch
3. Timing

To rule out case 1, get the strip to work with an Arduino.

To rule out case 2, take any strips that work on Arduino and try them on BB with a level converter.

If then you are left with only case 3 possible then I can make you a version with different timing, but looking at the data sheets for both 2811 and 2815, the timings seem to be within the specs of the standard 2812B timings (assuming the WS2811 is running in high speed mode. 

LMK what you find!

jammerxd, 2021-01-19:
Hi @bigjosh ,

Is there a link you can give to me of a pre-assembled logic level shifter that requires no soldering?

That's the last part I'm stuck on as the WS2811 strips work fine on an arduino and a raspberry pi 3 model b+ when hooked up to data directly.

Thanks

bigjosh, 2021-01-19:
The Pi is also only 3.3V, so if your strips work on it then it is not a voltage issue. 

Can you send me a git with the code (including any settings and `#defines`) you are using on the Arduino? I can then run it on an Arduino here and capture the exact timings it is generating and make a `.p` module to reproduce them on the BBB.

jammerxd, 2021-03-23:
Hi @bigjosh 

Sorry I didn't see this until now,

Here's the arduino code I'm using:

https://gitlab.com/jammerxd/ledspectrumanalyzer/-/blob/master/arduino/music_led_mirror/music_led_mirror.ino

I'm actually using the WS2811 timings. I think it's just because this package requires such an out of date debian package I've had such trouble even getting this to setup correctly. 

The debian packages for wheezy don't even exist anymore so I can't get updates and install new packages - including vncserver and htop - the animations even for a single strand at 150 pixels is 5 fps, not the 80 showed in the output.

```
root@beaglebone:~# apt-get update

Hit http://repos.rcn-ee.com wheezy Release.gpg
Ign http://security.debian.org wheezy/updates Release.gpg
Ign http://security.debian.org wheezy/updates Release
Hit http://repos.rcn-ee.com wheezy Release
Ign http://httpredir.debian.org wheezy Release.gpg
Ign http://httpredir.debian.org wheezy-updates Release.gpg
Get:1 http://repos.rcn-ee.com wheezy/main armhf Packages [124 kB]
Ign http://httpredir.debian.org wheezy Release
Ign http://httpredir.debian.org wheezy-updates Release
Err http://security.debian.org wheezy/updates/main armhf Packages
  404  Not Found [IP: 151.101.66.132 80]
Err http://security.debian.org wheezy/updates/contrib armhf Packages
  404  Not Found [IP: 151.101.66.132 80]
Err http://security.debian.org wheezy/updates/non-free armhf Packages
  404  Not Found [IP: 151.101.66.132 80]
100% [Waiting for headers]
Err http://httpredir.debian.org wheezy/main armhf Packages
  404  Not Found [IP: 199.232.30.132 80]
Err http://httpredir.debian.org wheezy/contrib armhf Packages
  404  Not Found [IP: 199.232.30.132 80]
Err http://httpredir.debian.org wheezy/non-free armhf Packages
  404  Not Found [IP: 199.232.30.132 80]
Err http://httpredir.debian.org wheezy-updates/main armhf Packages
  404  Not Found [IP: 199.232.30.132 80]
Err http://httpredir.debian.org wheezy-updates/contrib armhf Packages
  404  Not Found [IP: 199.232.30.132 80]
Err http://httpredir.debian.org wheezy-updates/non-free armhf Packages
  404  Not Found [IP: 199.232.30.132 80]
Fetched 1 B in 6s (0 B/s)
W: Failed to fetch http://security.debian.org/dists/wheezy/updates/main/binary-armhf/Packages  404  Not Found [IP: 151.101.66.132 80]

W: Failed to fetch http://security.debian.org/dists/wheezy/updates/contrib/binary-armhf/Packages  404  Not Found [IP: 151.101.66.132 80]

W: Failed to fetch http://security.debian.org/dists/wheezy/updates/non-free/binary-armhf/Packages  404  Not Found [IP: 151.101.66.132 80]

W: Failed to fetch http://httpredir.debian.org/debian/dists/wheezy/main/binary-armhf/Packages  404  Not Found [IP: 199.232.30.132 80]

W: Failed to fetch http://httpredir.debian.org/debian/dists/wheezy/contrib/binary-armhf/Packages  404  Not Found [IP: 199.232.30.132 80]

W: Failed to fetch http://httpredir.debian.org/debian/dists/wheezy/non-free/binary-armhf/Packages  404  Not Found [IP: 199.232.30.132 80]

W: Failed to fetch http://httpredir.debian.org/debian/dists/wheezy-updates/main/binary-armhf/Packages  404  Not Found [IP: 199.232.30.132 80]

W: Failed to fetch http://httpredir.debian.org/debian/dists/wheezy-updates/contrib/binary-armhf/Packages  404  Not Found [IP: 199.232.30.132 80]

W: Failed to fetch http://httpredir.debian.org/debian/dists/wheezy-updates/non-free/binary-armhf/Packages  404  Not Found [IP: 199.232.30.132 80]

E: Some index files failed to download. They have been ignored, or old ones used instead.
root@beaglebone:~#
```

`root@beaglebone:~/LEDscape-yona2# ./opc-server --no-lut --no-dithering --no-interpolation --count 150 --strip-count 1`

bigjosh, 2021-03-25:
Maybe try this ready-to-run image from my repo...

https://github.com/bigjosh/De-Harak-Clock-Controller/releases/tag/1.0BB

Put it on an SD card, then burn that into a BBB.  Connect pixel strings to any or all of these pins....

[image]

...and see what you see.

Note I can do `apt-get update` in this image and it seems to work fine.  Root user is `root`, no password.

jammerxd, 2021-03-25:
Thanks,

I'll give this a try - just out of curiosity, is there a way to have there be more than 7 channels?

Josh

bigjosh, 2021-03-25:
>is there a way to have there be more than 7 channels?

Yes, you can edit the `ledscape.config` after you successfully test with channels 0-7. The purpose of the test is to eliminate possible sources of problems, so initially use those channels and do to make any changes to the image config. 

You should expect to see the `fade` demo on any LEDs connected to those pins after the BBB boots up.

jammerxd, 2021-03-25:
Trying this now - I'll let you know! Thank you so much!

jammerxd, 2021-03-25:
So none of those channels produced a demo fading pattern

jammerxd, 2021-03-25:
I ran my own demo c# code and it looks like it can handle the FPS I want now. I don't know why there were so many issues before. Thank you so much!

jammerxd, 2021-03-25:
Can you update that image with a full mapping of all the channels and any other steps needed to enable them? I know I can go into the ledscape.config file and adjust the led strip count but I don't know what the other channels are mapped to due to conflicting data.

bigjosh, 2021-03-26:
The pin mapping scheme is set with the `outputMapping` param in the `/etc/ledscape-config.json`.  You can see the scheme mapping files in `pru/mappings/*.json`.  You can also print a channel-to-pin map using the command `node pru/pinmap.js --mapping <mapping-id>` where `<mapping-id>` is the mapping ID you want to print.

bigjosh, 2021-03-26:
https://github.com/bigjosh/LEDscape#pin-mappings

jammerxd, 2021-03-30:
Thanks @bigjosh for helping with this! Looks like everything is all set to be expanded! WS2815 and WS2812 strips work fine, I'm just wondering what I'm supposed to do if I have varying lengths of LED strips per channel and if that's something that's even supported?

bigjosh, 2022-07-13:
@nargetdev Hmmm... there are several things that could prevent ledscape from getting to some of the pins. There is all of the devicetree stuff which can be a pain, and the HDMI stuff too which I think requires out-of-band modifications to files that are not in the normal file system. I am away from my BBB for a couple of weeks, but those are the first places I would look. Try to map which pins are working and which are not (pretty easy using a pixel strip or oscilloscope ) and try to notice the pattern, and then see fi you can manually "fix" one of the not working pins by manual messing with its devicetree or pinmode. Report back what you find!

nargetdev, 2022-07-13:
[image]

fwiw here is an image of the pins that are working.  The *blue* pins are working.  Ignore the red (my PCB haha)

nargetdev, 2022-07-13:
I see this among the logs at boot time:

`uboot_overlays: [uboot_base_dtb=am335x-boneblack-uboot-univ.dtb]`

Does this mean it chose the wrong dtb or can there be multiple loaded at different times (i.e. does ledscape.service set the dtb at runtime?)

bigjosh, 2022-07-13:
Completely aside, if you are making a custom PCB, I'd strongly suggest connecting a free GPIO to the reset pin if you can so you will have a hardware way to deal with the horrible dead-ethernet port problem on BBB...

https://wp.josh.com/2018/06/04/a-software-only-solution-to-the-vexing-beagle-bone-black-phy-issue/