Yona-Appletree/LEDscape #40: New PRU support: rpmsg?

JetForMe, 2015-10-27:
It seems TI has broken uio_pruss in favor of its new PRU RPM stuff, starting with RCN's 4.1.11-ti kernels. Recently John Syne [patched](https://github.com/RobertCNelson/ti-linux-kernel-dev/blob/393f5ea242a86544063ec2af63ef270094bfa1da/patches/pru/0003-Add-rpmsg_pru-support.patch) the BBB kernel to fix some issues. I wonder if LEDscape will be updated to use this new mechanism?

Yona-Appletree, 2015-11-02:
I personally don't have plans to make that upgrade for the time being, but I am open to pull requests.

JetForMe, 2015-11-03:
Okay, I'll investigate how to use it.

JetForMe, 2015-11-03:
I just thought you (or maybe I should mention to the original authors) might want to make sure it works in the newer kernels.

Yona-Appletree, 2015-11-03:
If you have the bandwidth, I'm absolutely in favor of having compatibility with the new communication scheme, and would welcome documentation, research, POCs or full-on pull requests you have.