Yona-Appletree/LEDscape #42: Use more than 64 Pixel with Processing

gsandreas, 2015-12-10:
Hi, 

I want to use the BBB for powering in total 50Meters of LED stripes, seperatated in 10 parts with different lenghts which are not arragend in a Matrix. The longest has a total amount of 720 single LEDs. 

I've worked with Processing and figured out, that there is a 64 LED limitation, which comes from the fadecandy implemtation. Is there any way, to use all capabilities (more than 64 LEDs per Channel) of the BBB with Processing?

Furthermore i made some experiments with shell comands to the opc - server and i could not get it to work. It shows always the demo pattern which i can not change. 

You mentioned that there is a documentation available. Do you mean your readme.md in github? Unfortunatly i could only get the BBB to work with the demo pattern, and with your readme.md i could not find the failure i made. 

Regards 
Andreas

LightAtPlay, 2015-12-31:
Could you give us more information about the problems you've had, especially with launching opc-server itself? The demo pattern is shown when no input data is coming in via OPC, though it can be disabled. Please look at the help outputted from opc-server (-h) for more information about the various flags.

Given you want to drive 720 LEDs per strip, you probably want an opc invocation like this:

    ./opc-server --count 64 --mode ws281x

I am not aware of a 64 pixel limit in the processing OPC library, but it is worth mentioning that it has been adapted from Fadecandy which is designed for for 64-led strips, so there might be an issue. I'll take a look later when I have access to my test hardware.

Daniel-dk, 2016-09-05:
We've used it for a few thousand LEDs, 
make sure that the OPC server on BBB is set up to expect the number of LEDs in your system ( led count , number of outputs etc)

and AFAIK, there isnt much change on the processing side other than making sure you send the right number of pixels the OPC server expects

Yona-Appletree, 2016-09-12:
I'm going to close this issue as I think it's something specific to the OP's environment. I'm aware that better docs and a simpler setup process would be good, but I think that's beyond the slip of this issue. @gsandreas please contact me directly, lightatplay@gmail.com, if you have any further questions, I would be glad to help, given the time.

closed 2016-09-12 by Yona-Appletree