Yona-Appletree/LEDscape #43: PRU initialization on current build of Debian Jessy, kernel 4.1

GGreenwood, 2016-04-22:
The latest installation docs don't work for new installations of Debian Jessy(tested with the 2016-01-24 image). Not only is the dtbs location different(which the docs do cover), but TI introduced a new version of the 4.1 kernel(see [this](http://stackoverflow.com/a/35368229/3469536)) which changes the API for accessing the PRUSS. This causes the `modprobe uio_pruss` step to fail.

The solution is to execute the following before copying the modified `am335x-boneblack.dtb` file:
```
cd /opt/scripts/tools
./update_kernel.sh --bone-rt-kernel --lts-4_1
reboot
```

However, from this point on, I receive segmentation faults when running LEDscape.

```
root@ledscape:~/LEDscape# ./run-ledscape 
tokens: 381: {
	"outputMode": "ws281x",
	"outputMapping": "rgb-123-v2",
	"demoMode": "power",
	"ledsPerStrip": 30,
	"usedStripCount": 23,
	"colorChannelOrder": "BRG",
	"opcTcpPort": 7890,
	"opcUdpPort": 7890,
	"enableInterpolation": true,
	"enableDithering": true,
	"enableLookupTable": true,
	"lumCurvePower": 2.0000,
	"whitePoint": {
		"red": 0.9000,
		"green": 1.0000,
		"blue": 1.0000
	}
}

Loaded config file from ws281x-config.json.
Config file written to ws281x-config.json
[main] Starting server on ports (tcp=7890, udp=7890) for 30 pixels on 48 strips
[main] Demo Mode Enabled
Starting demo data thread
[e131] Starting UDP server on port 5568
[udp] Starting UDP server on port 7890
Segmentation fault
```

bigjosh, 2016-09-04:
I'd be happy to sponsor work to either document a workaround with current code base, or move the code to the new  remoteproc driver if anyone is interested.

Yona-Appletree, 2016-09-12:
@bigjosh I'll try to find time to address this. Do you know if this image of Debian is shipping with current BBBs?

Yona-Appletree, 2016-09-12:
@bigjosh Please feel free to contact me at lightatplay@gmail.com to discuss the issue more, and potentially to sponsor this work.

bigjosh, 2016-09-12:
@Yona-Appletree Good question- I always re-image boards as soon as they come in, but my guess is that no one is shipping with the new kernel because we'd be seeing more people asking about the problems.

bigjosh, 2016-09-12:
BTW, don't waste too much time looking at the `remoteproc` stuff if you aren't familiar with it yet. I've already fallen back to the old kernel because of other unrelated problems on the new distributions so this work could likely be postponed for a while without bothering anyone.

steely-glint, 2016-10-29:
Did you find a work around for the segfaults ? 
I am using bone-debian-8.4-iot-armhf-2016-05-13-4gb.img 
and the bone kernel as described above. 
(I need the newer kernel because it supports tethering from a cheap android phone on the USB port).
I'd be happy to help debug if I'm given hints.....

mykolasmith, 2016-11-27:
Can someone point me in the direction of a Debian distro that does work? I am also getting `Segmentation fault` when trying to run LEDscape on the latest official image(s).

GGreenwood, 2016-11-27:
@mykolasmith I believe that I got it working by going back one Debian release. Try the "Debian 7.11 2015-06-15 4GB SD LXDE" release [here](https://beagleboard.org/latest-images).

mykolasmith, 2016-11-27:
@GGreenwood Thanks for the quick response. I'll try that =)

steely-glint, 2016-11-27:
Here are my build notes on using the latest distro: 
https://github.com/steely-glint/starway/blob/novparkreno/bbbuildnotes.txt
It takes a bit of a dance - but it seems pretty stable - that's been running for a couple of weeks driving starway