Yona-Appletree/LEDscape #53: Compilation failure

Ashoat, 2017-04-17:
Hi there! First of all, thanks so much for supporting this project.

I'm trying to compile and install the LEDscape project as per instructions. I have a Beaglebone Green running the latest Wheezy 7.11 Debian build. I've checked out the LEDscape project, set the executable bit on the `install-software.sh` script, and attempted to run the script. It gets the following failure when running the `make` command:

```
debian@beaglebone:~/LEDscape$ sudo ./install-software.sh
Making ledscape...
gcc -std=c99 -W -Wall -D_DEFAULT_SOURCE -Wp,-MMD,./.opc-server.o.d -Wp,-MT,opc-server.o -I. -O2  -g -lm -mtune=cortex-a8 -march=armv7-a -Wunused-parameter -DNS_ENABLE_IPV6 -Wsign-compare -Werror -Wno-unknown-pragmas -I./am335x/app_loader/include -c -o opc-server.o opc-server.c
In file included from opc-server.c:27:0:
lib/cesanta/frozen.h:38:3: error: unknown type name 'uint'
opc-server.c: In function 'demo_mode_from_string':
opc-server.c:146:2: error: implicit declaration of function 'strcasecmp' [-Werror=implicit-function-declaration]
opc-server.c: In function 'main':
opc-server.c:719:2: error: implicit declaration of function 'bzero' [-Werror=implicit-function-declaration]
opc-server.c: In function 'server_config_from_json':
opc-server.c:1018:59: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
opc-server.c:1018:59: error: signed and unsigned type in conditional expression [-Werror=sign-compare]
opc-server.c:1022:36: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
opc-server.c:1022:36: error: signed and unsigned type in conditional expression [-Werror=sign-compare]
opc-server.c:1027:36: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
opc-server.c:1027:36: error: signed and unsigned type in conditional expression [-Werror=sign-compare]
opc-server.c:1032:36: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
opc-server.c:1032:36: error: signed and unsigned type in conditional expression [-Werror=sign-compare]
opc-server.c:1037:36: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
opc-server.c:1037:36: error: signed and unsigned type in conditional expression [-Werror=sign-compare]
opc-server.c:1042:36: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
opc-server.c:1042:36: error: signed and unsigned type in conditional expression [-Werror=sign-compare]
opc-server.c:1047:36: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
opc-server.c:1047:36: error: signed and unsigned type in conditional expression [-Werror=sign-compare]
opc-server.c:1052:36: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
opc-server.c:1052:36: error: signed and unsigned type in conditional expression [-Werror=sign-compare]
opc-server.c:1057:36: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
opc-server.c:1057:36: error: signed and unsigned type in conditional expression [-Werror=sign-compare]
opc-server.c:1062:36: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
opc-server.c:1062:36: error: signed and unsigned type in conditional expression [-Werror=sign-compare]
opc-server.c:1067:36: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
opc-server.c:1067:36: error: signed and unsigned type in conditional expression [-Werror=sign-compare]
opc-server.c:1072:36: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
opc-server.c:1072:36: error: signed and unsigned type in conditional expression [-Werror=sign-compare]
opc-server.c:1077:36: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
opc-server.c:1077:36: error: signed and unsigned type in conditional expression [-Werror=sign-compare]
opc-server.c:1082:36: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
opc-server.c:1082:36: error: signed and unsigned type in conditional expression [-Werror=sign-compare]
opc-server.c: In function 'rotate_frames':
opc-server.c:1286:3: error: implicit declaration of function 'timersub' [-Werror=implicit-function-declaration]
opc-server.c: In function 'render_thread':
opc-server.c:1334:4: error: implicit declaration of function 'usleep' [-Werror=implicit-function-declaration]
opc-server.c: In function 'join_multicast_group_on_all_ifaces':
opc-server.c:1782:29: error: 'IFF_LOOPBACK' undeclared (first use in this function)
opc-server.c:1782:29: note: each undeclared identifier is reported only once for each function it appears in
opc-server.c:1783:29: error: 'IFF_POINTOPOINT' undeclared (first use in this function)
opc-server.c:1784:28: error: 'IFF_MULTICAST' undeclared (first use in this function)
opc-server.c:1787:19: error: storage size of 'multicast_req' isn't known
opc-server.c:1787:19: error: unused variable 'multicast_req' [-Werror=unused-variable]
opc-server.c: In function 'e131_server_thread':
opc-server.c:1905:7: error: 'uint' undeclared (first use in this function)
cc1: all warnings being treated as errors
make: *** [opc-server.o] Error 1
Making backups of old device tree files...
mkdir: cannot create directory `/boot/dtbs/3.8.13-bone80/ledscape.bak': File exists
Copying new device tree files...
Copying config file to /etc
Leaving existing /etc/ledscape-config.json intact.
Enabling kernel module...
Done. Please enter reboot to reboot the machine and enable changes.
debian@beaglebone:~/LEDscape$
```

Any idea what's going on? Should I not be relying on the stock compiler packaged with the BBB Debian images?

Would appreciate any advice. Thanks!

Yona-Appletree, 2017-04-17:
I'll have to look into this. To be honest, it's hard for me to keep up with all the different combinations of board revision, kernel version, bootloader setup, device trees and compiler versions. I have an LEDscape Arch-based image that I think I'm going to release soon so we can have a single, non-moving target for LEDscape. Stay tuned, and I'm happy to give you a copy via email.

GGreenwood, 2017-04-17:
Count me interested in testing the Arch-based image. I've got LEDscape driving a 23x30 display right now.

Ashoat, 2017-04-17:
Figured it out! Sorry for the bother. It looks like the "Debian 7.11 2015-06-15 4GB SD LXDE" image in the instructions has an older glibc version (2.13) than the one the Makefile is configured for (`_DEFAULT_SOURCE` replaces `_BSD_SOURCE` in 2.19). The following simple patch to `Makefile` should make it work with the stock glibc on the image:

```
--- a/Makefile
+++ b/Makefile
@@ -31,6 +31,7 @@ CFLAGS += \
        -W \
        -Wall \
        -D_DEFAULT_SOURCE \
+       -D_BSD_SOURCE \
        -Wp,-MMD,$(dir $@).$(notdir $@).d \
        -Wp,-MT,$@ \
        -I. \
```

Probably an alternate approach would be to add a step to the instructions to update glibc before compilation

closed 2017-04-18 by Ashoat

anthonyn, 2017-08-06:
Thanks for posting this, I hit the same thing!