Yona-Appletree/LEDscape #56: install-software.sh failing?

voidnoise, 2017-06-13:
Hi, 

I just did a clean build of Debian 7.11 2015-06-15 4GB SD LXDE 

First I install
```
sudo apt-get update
sudo apt-get install usbmount -y
sudo apt-get install git build-essential python -y
```

I then follow the read.me instructions
```
git clone git://github.com/Yona-Appletree/LEDscape
cd LEDscape
chmod +x install-software.sh
./install-software.sh
reboot
```

The output of install-software.sh is as below;
```
Making ledscape...
gcc -std=c99 -W -Wall -D_DEFAULT_SOURCE -Wp,-MMD,./.opc-server.o.d -Wp,-MT,opc-server.o -I. -O2  -g -lm -mtune=cortex-a8 -march=armv7-a -Wunused-parameter -DNS_ENABLE_IPV6 -Wsign-compare -Werror -Wno-unknown-pragmas -I./am335x/app_loader/include -c -o opc-server.o opc-server.c
In file included from opc-server.c:27:0:
lib/cesanta/frozen.h:38:3: error: unknown type name ‘uint’
opc-server.c: In function ‘demo_mode_from_string’:
opc-server.c:146:2: error: implicit declaration of function ‘strcasecmp’ [-Werror=implicit-function-declaration]
opc-server.c: In function ‘main’:
opc-server.c:719:2: error: implicit declaration of function ‘bzero’ [-Werror=implicit-function-declaration]
opc-server.c: In function ‘server_config_from_json’:
opc-server.c:1018:59: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
opc-server.c:1018:59: error: signed and unsigned type in conditional expression [-Werror=sign-compare]
opc-server.c:1022:36: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
opc-server.c:1022:36: error: signed and unsigned type in conditional expression [-Werror=sign-compare]
opc-server.c:1027:36: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
opc-server.c:1027:36: error: signed and unsigned type in conditional expression [-Werror=sign-compare]
opc-server.c:1032:36: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
opc-server.c:1032:36: error: signed and unsigned type in conditional expression [-Werror=sign-compare]
opc-server.c:1037:36: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
opc-server.c:1037:36: error: signed and unsigned type in conditional expression [-Werror=sign-compare]
opc-server.c:1042:36: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
opc-server.c:1042:36: error: signed and unsigned type in conditional expression [-Werror=sign-compare]
opc-server.c:1047:36: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
opc-server.c:1047:36: error: signed and unsigned type in conditional expression [-Werror=sign-compare]
opc-server.c:1052:36: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
opc-server.c:1052:36: error: signed and unsigned type in conditional expression [-Werror=sign-compare]
opc-server.c:1057:36: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
opc-server.c:1057:36: error: signed and unsigned type in conditional expression [-Werror=sign-compare]
opc-server.c:1062:36: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
opc-server.c:1062:36: error: signed and unsigned type in conditional expression [-Werror=sign-compare]
opc-server.c:1067:36: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
opc-server.c:1067:36: error: signed and unsigned type in conditional expression [-Werror=sign-compare]
opc-server.c:1072:36: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
opc-server.c:1072:36: error: signed and unsigned type in conditional expression [-Werror=sign-compare]
opc-server.c:1077:36: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
opc-server.c:1077:36: error: signed and unsigned type in conditional expression [-Werror=sign-compare]
opc-server.c:1082:36: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
opc-server.c:1082:36: error: signed and unsigned type in conditional expression [-Werror=sign-compare]
opc-server.c: In function ‘rotate_frames’:
opc-server.c:1286:3: error: implicit declaration of function ‘timersub’ [-Werror=implicit-function-declaration]
opc-server.c: In function ‘render_thread’:
opc-server.c:1334:4: error: implicit declaration of function ‘usleep’ [-Werror=implicit-function-declaration]
opc-server.c: In function ‘join_multicast_group_on_all_ifaces’:
opc-server.c:1782:29: error: ‘IFF_LOOPBACK’ undeclared (first use in this function)
opc-server.c:1782:29: note: each undeclared identifier is reported only once for each function it appears in
opc-server.c:1783:29: error: ‘IFF_POINTOPOINT’ undeclared (first use in this function)
opc-server.c:1784:28: error: ‘IFF_MULTICAST’ undeclared (first use in this function)
opc-server.c:1787:19: error: storage size of ‘multicast_req’ isn’t known
opc-server.c:1787:19: error: unused variable ‘multicast_req’ [-Werror=unused-variable]
opc-server.c: In function ‘e131_server_thread’:
opc-server.c:1905:7: error: ‘uint’ undeclared (first use in this function)
cc1: all warnings being treated as errors
make: *** [opc-server.o] Error 1
Making backups of old device tree files...
Copying new device tree files...
Copying config file to /etc
Leaving existing /etc/ledscape-config.json intact.
Enabling kernel module...
Done. Please enter reboot to reboot the machine and enable changes.
```
The make: *** [opc-server.o] Error 1 is a little worrying.

After reboot I run-ledscape and get the following output;
`./run-ledscape: 11: ./run-ledscape: ./opc-server: not found
`
I noted that a recent [commit](https://github.com/Yona-Appletree/LEDscape/commit/47d267c3a9a6a988356cbef6c9941d29bbdddb8b) was to fix a compilation error. Any chance its not quite fixed? Any suggestions on how to fix it?

voidnoise, 2017-06-13:
Update: I check out commit 976f1ddf264191736d9569977244a5b6be20e65a and it now compiles and run-ledscape runs successfully. Im guessing in the following commits something broke for my set up at least.

orenyomtov, 2019-09-01:
Thank you, 2 years later and this workaround is still needed