Yona-Appletree/LEDscape #69: Compile of opc-server.c fails on Debian 10.3 on BBB

pscheie, 2022-03-08:
I've got a Beaglebone Black running Debian 10.3.  When I run 'make' to build the opc-server, it's failing thusly:

opc-server.c: In function ‘opc_server_set_error.part.1’:
opc-server.c:211:3: error: invalid use of ‘__builtin_va_arg_pack ()’
   snprintf(
   ^~~~~~~~~
    extra_info_out,
    ~~~~~~~~~~~~~~~
    sizeof(extra_info_out),
    ~~~~~~~~~~~~~~~~~~~~~~~
    extra_info,
    ~~~~~~~~~~~
    __builtin_va_arg_pack()
    ~~~~~~~~~~~~~~~~~~~~~~~
   );

There are also a bunch of warnings beforehand.  Any suggestions on how to fix this?

DBell33, 2022-04-12:
having the same issue currently, anyone figured out a workaround to this issue?

nargetdev, 2022-07-13:
For whatever reason setting the compiler optimization from

-O2 --> -O1

suppresses the error... but you'll still have linker errors in the end.

just deleting the line and doing

     char extra_info_out[2048];
     snprintf(
             extra_info_out,
             sizeof(extra_info_out),
             extra_info
     );

allows to compile.. but I'm still getting a segmentation fault..

nargetdev, 2022-07-13:
FWIW ThrowingBagels works out of the box.. only caveat is new packet format.

https://github.com/iontank/ThrowingBagels

nargetdev, 2022-07-13:
Ok.. so for the segmentation faults check out here:

https://github.com/steely-glint/starway/blob/novparkreno/bbbuildnotes.txt

Specifically the recommendations as follows:

    vi opc-server.c
    change e131 port to 0 in
    +       .e131_port = 0,
    And
    +       assert_int_range_inclusive("e131 UDP Port", 0, 65535, input_config- 
    >e131_port);
    
    make

nargetdev, 2022-07-13:
Geocene has a working version in 4.19:    https://github.com/Geocene/LEDscape