Yona-Appletree/LEDscape #70: modernizing to 10.3 "flasher" image

nargetdev, 2022-07-13:
I'd like to get LEDscape running on a newer image than Wheezy.

I've gotten everything to compile, however I'm bumping into nonexistent GPIO at runtime as follows.. wonder if this is something easy to fix?  I notice that if i `ls` on the `/sys/class/gpio` dir I have loads of available in higher numbers, however the single digits are lacking.

Specifically the line: `pru_gpio:173: /sys/class/gpio/gpio2/value: Unable to open?`

note that `gpio2` above randomly changes between 2, 3, and 7 on different runs.

full message...

```
Loaded config file from /etc/ledscape-config.json.
Config file written to /etc/ledscape-config.json
[main] Starting server on ports (tcp=7890, udp=7890) for 20 pixels on 48 strips
[main] Demo Mode Enabled
Allocating buffers for 960 pixels (23040 bytes)
[main] Initializing / Updating server...frame_size1=960
Starting demo data thread
[e131] Not starting e131 server; Port is zero.
[render] Starting render thread for 960 total pixels
[udp] Starting UDP server on port 7890
[main] Starting LEDscape...pru_init: PRU 0: data 0xb456d000 @ 8192 bytes,  DMA 0xb44ed000 / 9c940000 @ 262144 bytes
pru_init: PRU 1: data 0xa4446000 @ 8192 bytes,  DMA 0xa42c0000 / 9c940000 @ 262144 bytes
pru_gpio:173: /sys/class/gpio/gpio2/value: Unable to open? No such file or directory
```

nargetdev, 2022-07-13:
here's  the `opc-server.c` file -- https://github.com/jamie-mint/LEDscape/blob/master/opc-server.c

nargetdev, 2022-07-13:
UPDATE: thankyou @Geocene  This works in 4.19 -->  https://github.com/Geocene/LEDscape

jammerxd, 2023-03-11:
Does this mean that the version you linked works with the current image?